ros/pluginlib #43: Pluginlib 1.10.4 not released?

davetcoleman, 2016-09-25:
The latest shadow-fixed Kinetic Xenial release is still using pluginlib 1.10.3, even though there was just a sync and bloom says it should be [1.10.4](https://github.com/ros/rosdistro/blob/master/kinetic/distribution.yaml#L3167) - why is that? I confirmed this both in Docker containers and locally. The Travis build cannot pass until the latest plugin lib is available.

tfoote, 2016-09-25:
There's a major rebuilds going on at the moment. Once the rebuild is complete it will sync to shadow-fixed. 

[1.10.4 is in building](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=pluginlib) at the moment. 

There's some 2000+ jobs in the queue still down from 5k. http://build.ros.org:8080/view/Queue/ This is a big rebuild with a patch to catkin. We're scaled up to 24 executors 24 hours ago then 44 executors today. [[the load]]
(http://build.ros.org:8080/label/buildslave/load-statistics?type=min) It will be probably another 12+ hours before all architectures sync.

closed 2016-09-25 by tfoote

davetcoleman, 2016-09-25:
Ah I hadn't considered build farm lag, thanks for the explanation!