ros/pluginlib #44: Thread Safety Issues

Jmeyer1292, 2016-10-03:
I want to open with a thank you to the maintainers of these core packages. I hope I'm not too obnoxious in my questioning.

I have recently written an application that crunches numbers in a distributed fashion and uses ROS plugins to help do some of that work. I am running into issues that appear to be related to the threading I'm doing. Specifically, I am:
 * spawning **I** threads
 * each thread then creates **J** instances of a particular plugin
 * these threads then stay alive and accept work through another interface

What I'm finding is that with sufficiently high contention, on ROS Indigo, a plugin will eventually fail to load. With a single thread, or mutex protected access it works fine consistently.

I have modified the pluginlib tutorial code to provide a minimal example of this bug:
```c++
#include <iostream>
#include <boost/shared_ptr.hpp>
#include <boost/thread.hpp>

#include <pluginlib/class_loader.h>
#include <pluginlib_tutorials/polygon_base.h>
#include <boost/ref.hpp>

typedef boost::shared_ptr<polygon_base::RegularPolygon> RegularPolygonPtr;

static RegularPolygonPtr loadPlugin(pluginlib::ClassLoader<polygon_base::RegularPolygon>& loader)
{
  try
  {
    RegularPolygonPtr plugin = loader.createInstance("pluginlib_tutorials/regular_triangle");
    return plugin;
  }
  catch (std::exception& e)
  {
    std::cerr << "An error occured: " << e.what() << "\n";
    return RegularPolygonPtr();
  }
}

// Experiment 1: Each worker has its own class loader
void workerThread1(int n)
{
  pluginlib::ClassLoader<polygon_base::RegularPolygon> loader ("pluginlib_tutorials", "polygon_base::RegularPolygon");
  std::vector<RegularPolygonPtr> plugins;

  for (int i = 0; i < n; ++i)
  {
    RegularPolygonPtr plugin = loadPlugin(loader);
    if (plugin) plugin->initialize(10.0);
    plugins.push_back(plugin);
  }

  boost::this_thread::sleep(boost::posix_time::milliseconds(1000));
}

void launchExperiment1(int n_threads, int n)
{
  std::vector<boost::thread*> threads;
  for (int i = 0; i < n_threads; ++i)
  {
    threads.push_back( new boost::thread(workerThread1, n) );
  }

  for (int i = 0; i < n_threads; ++i)
  {
    threads[i]->join();
    delete threads[i];
  }
}

// Experiment 2: All workers share a class loader
void workerThread2(int n, pluginlib::ClassLoader<polygon_base::RegularPolygon>& loader)
{
  std::vector<RegularPolygonPtr> plugins;

  for (int i = 0; i < n; ++i)
  {
    RegularPolygonPtr plugin = loadPlugin(loader);
    if (plugin) plugin->initialize(10.0);
    plugins.push_back(plugin);
  }

  boost::this_thread::sleep(boost::posix_time::milliseconds(1000));
}

void launchExperiment2(int n_threads, int n)
{
  pluginlib::ClassLoader<polygon_base::RegularPolygon> loader ("pluginlib_tutorials", "polygon_base::RegularPolygon");
  std::vector<boost::thread*> threads;
  for (int i = 0; i < n_threads; ++i)
  {
    threads.push_back( new boost::thread(workerThread2, n, boost::ref(loader)) );
  }

  for (int i = 0; i < n_threads; ++i)
  {
    threads[i]->join();
    delete threads[i];
  }
}

int main(int argc, char** argv)
{
  if (argc < 3)
  {
    std::cerr << "Usage: ./polygon_loader <n_threads> <n_items_per_thread>";
    return 1;
  }

  int n_threads = std::atoi(argv[1]);
  int n_items_per_thread = std::atoi(argv[2]);

  std::cout << "Starting multi-threaded loading experiment w/ " << n_threads << " threads " <<
            "and " << n_items_per_thread << " plugins per thread" << std::endl;

  std::cout << "Experiment 1: All threads have own class loader" << std::endl;
  launchExperiment1(n_threads, n_items_per_thread);
  std::cout << "Experiment 2: All threads share a class loader" << std::endl;
  launchExperiment2(n_threads, n_items_per_thread);
  return 0;
}
```

If you build this in place of the existing ```polygon_loader.cpp``` file in [this](https://github.com/ros/common_tutorials/tree/hydro-devel/pluginlib_tutorials) package you can reproduce the issue. On my machine, you'll see errors with ```rosrun polygon_loader 100 500``` pretty consistently.

Is this a known issue? Does the pluginlib make any guarantees about thread safety (I see that the separate class_loader package says its thread safe)?

Synchronously loading plugins is not a very viable solution. If you need more information, I'm happy to help provide it.

jrgnicho, 2016-10-04:
@Jmeyer1292 I ran your example and indeed it kept throwing __CreateClassException__ exceptions almost every time.  Adding a mutex at the beginning of the `loadPlugin` function fixed it but it took 8.5 seconds  on average for all the experiments to complete with 200 threads and 200 plugins for each.

Jmeyer1292, 2016-10-05:
@jrgnicho @mikaelarguedas 

I think I've tracked down _an_ issue that could be partially responsible for the issues I am seeing. It involves a logical & data race condition in _class_loader core's_ underlying ```loadLibrary``` function [here](https://github.com/ros/class_loader/blob/indigo-devel/src/class_loader_core.cpp#L415).

I'll open up a discussion over there to further that point.

Pluginlib also appears to have the potential for data races under multi-threaded conditions. Unlike class_loader, the documentation makes no assurances about thread safety however. See [this](https://github.com/ros/pluginlib/blob/indigo-devel/include/pluginlib/class_loader_imp.h#L558) line for an example.

jrgnicho, 2016-10-05:
@Jmeyer1292 I can see how the logic in the `loadLibrary` function could allow for race conditions.  Perhaps the appropriate fix is to move the [mutex locking line ](https://github.com/ros/class_loader/blob/indigo-devel/src/class_loader_core.cpp#L432) to the beginning of the function.  Thanks for looking into this.

mikaelarguedas, 2016-10-18:
this should be fixed by https://github.com/ros/class_loader/issues/40.
Closing this for now. Feel free to reopen it if there are some corner cases where the crash still happens

closed 2016-10-18 by mikaelarguedas