ros/pluginlib #45: release createUniqueInstance into indigo

v4hn, 2016-10-17:
@mikaelarguedas mentioned in https://github.com/ros/pluginlib/pull/42#issuecomment-248457801 that the new `createUniqueInstance` interface will be released into indigo and jade too.

It would be great to see this happen in the near future!
MoveIt's main development branch (`kinetic-devel`) requires this feature.
In order to use the branch with ROS indigo, currently one has to overlay the /opt/ros/indigo version of `pluginlib` in the workspace.
It is not easy to upgrade many robotic setups, so this kind of cross-versioning is not too uncommon.

mikaelarguedas, 2016-10-30:
Closing this now that the `createUniqueInstance` interface has been released into [Jade](http://repositories.ros.org/status_page/ros_jade_default.html?q=pluginlib) and [Indigo](http://repositories.ros.org/status_page/ros_indigo_default.html?q=pluginlib). It's currently in the testing repo (shadow-fixed) and will be available in the main repo after the next sync.

closed 2016-10-30 by mikaelarguedas

v4hn, 2016-10-31:
Awsome, thanks!