ros/pluginlib #51: Use provide support for std::shared_ptr

jlack1987, 2017-05-18:
I feel like somebody must have asked this already, but I was unable to find the issue if they have, so i'll ask. Are there any plans to phase out the `boost::shared_ptr` in favor of `std::shared_ptr`?

tfoote, 2017-05-18:
This cannot be done until development is only targeting C++11 minimum targets: http://www.ros.org/reps/rep-0003.html  Since boost is a integral part of the ROS1 API it's quite likely that for stability we will not significantly change this API.

In ROS2 we have switched to use C++11 instead of boost for core APIs.

mikaelarguedas, 2017-05-25:
As @tfoote mentioned, it's unlikely that the core of ROS1 will get rid of boost given that it'll break API for most of the ROS packages. I'll thus mark this as wontfix

mikaelarguedas, 2018-01-30:
@jlack1987 as the next version of ROS ([Melodic](https://github.com/ros-infrastructure/rep/pull/139/files#diff-41ae062315c36c45afaa5c192b2ca7c5R379)) will allow the use of c++14 types/features in ROS packages even in desktop-full, this is something that we can reconsider.

This will need to be additional API and keep providing the existing boost API to not break downstream packages consuming the existing API.
I'll remove the `wontfix` label for now but don't know if/when I'll have spare cycles to implement it.

caioaamaral, 2020-01-13:
Any updates about it? I saw that other packages (aka ros-controls/ros_controllers) has started shared pointers migration to std.

clalancette, 2021-05-14:
At this point, I don't think it is worth it to change this for ROS 1 version of pluginlib.  We've already moved away from boost for ROS 2.  As such, I don't think there is anything to do here so I'll close this out.

closed 2021-05-14 by clalancette