ros/pluginlib #86: The node with pluginlib can not run in custom boost situation

echotor, 2017-11-13:
The thing is like this:

I compile the boost of version 1.65 into my computer. And I build a package with pluginlib in it. I compile it over. But when I run the node, it will result in a segment fault. Like this:

```
gdb ./polygon_loader 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./polygon_loader...(no debugging symbols found)...done.
(gdb) r
Starting program: /home/wangyawei/RosWorkspace/common_tutorials_ws/devel/lib/pluginlib_tutorials/polygon_loader 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x2) at malloc.c:2951
2951    malloc.c: No such file or directory.
```

So the problem is segment fault.
When I remove the path /usr/local/include/boost and /usr/local/lib/libboost* and recompile the package and the error is gone.

So I check the version of the two kind of boost: 1.58 for origin ros and 1.65 for my compile. Maybe there are different point between two kinds version.

But I want to know could anyone update the pluginlib for the latest version of boost?

mikaelarguedas, 2017-11-13:
Hi @echotor , thanks for the report,

Could you provide a backtrace ? the current error "malloc.c: No such file or directory." doesnt provide much information as of where the program is crashing.

While I haven't tried boost 1.65, the version of pluginlib on the kinetic-devel works with boost 1.62 as it is the one used by most platforms in ROS Lunar. If there is anything to modify to work with 1.65 without breaking backward compatibility I'm happy to integrate the change.
Please consider providing either more information about what boost calls are crashing or (even better :wink:) a PR adding support for newer boost (in either case a reproducible example would be great)

mikaelarguedas, 2018-09-14:
Closing due to no response.

@echotor FYI pluginlib has been released in ROS Melodic that uses boost 1.65.1, you can check the code on the melodic-devel branch and check if this is still an issue for you. If yes feel free to comment here and this issue can be reopened

closed 2018-09-14 by mikaelarguedas