ros/pluginlib #102: ROS 2: erase superfluous semicolon

Martin-Idel-SI, 2018-03-13:
I have a problem when compiling object files using the pluginlib, when I get the following error:
`/include/pluginlib/class_list_macros.hpp:49:59: error: extra ‘;’ [-Werror=pedantic]
   CLASS_LOADER_REGISTER_CLASS(class_type, base_class_type);`

It's claiming that one should remove the last ";" in this line: 
https://github.com/ros/pluginlib/blob/b456beba88564b746694dde68696d8bc4edc78c9/pluginlib/include/pluginlib/class_list_macros.hpp#L49
That makes sense, since macros shouldn't end with a semicolon and indeed, everything still seems to work fine without it and I don't get the above error.

I sadly can't provide a PR at the moment.

mikaelarguedas, 2018-03-13:
Thanks @Martin-Idel-SI for pointing it out. #103 should address the issue

mikaelarguedas, 2018-03-13:
Fixed in #103 
Closing this manually as github doesn't seem to close linked issues automatically today

PR #103 (2018-03-13): ROS 2 : remove extra semicolon to fix pedantic warning
fixes #102 

* Linux [image]
* Linux-aarch64 [image]
* macOS [image] (unrelated test failure)
* Windows [image] (unrelated test failure)
closed 2018-03-13 by mikaelarguedas