ros/pluginlib #109: New compiler warning

dirk-thomas, 2018-04-02:
As mentioned in ros-visualization/qt_gui_core#114.

E.g. http://build.ros.org/job/Mpr__qt_gui_core__ubuntu_bionic_amd64/1/warnings22Result/

mikaelarguedas, 2018-04-03:
It's unclear to me why this shows up on this package. This function should be called transitively by createInstance.
<details>

Call stack being:
```
catkinFindLib
called by getCatkinLibraryPaths (l323)
called by getAllLibraryPathsToTry ('330)
called by getClassLibraryPath (l412)
called by loadLibraryForClass (608)
called by:
  -> createClassInstance (l129)
  -> createInstance (l158)
  -> createUniqueInstance (l190)
  -> createUnmanagedInstance (l229)

```
</details>

I built the melodic version of class_loader, pluginlib, qt_gui_core and rqt_gui with this additional print

Debug output of `rosrun rqt_gui rqt_gui`:
```
[DEBUG] [1522718065.187724347]: XML file specifies lookup name (i.e. magic name) = rqt_gui_cpp/RosCppPluginProvider.
[DEBUG] [1522718065.187735353]: Exiting determineAvailableClasses()...
[DEBUG] [1522718065.187742513]: Finished constructring ClassLoader, base = qt_gui_cpp::PluginProvider, address = 0x118eb80
[DEBUG] [1522718065.187829905]: Attempting to create managed instance for class rqt_gui_cpp/RosCppPluginProvider.
[DEBUG] [1522718065.187843431]: Class rqt_gui_cpp/RosCppPluginProvider maps to library lib/librqt_gui_cpp in classes_available_.
[DEBUG] [1522718065.187850493]: this function is actually called
```

@dirk-thomas can you please do the same experiment on a bionic system and see if that print appears?

dirk-thomas, 2018-04-04:
Same here - I don't see any differences.

I would suggest to just move the code from the anonymous `catkinFindLib` function into the only function calling it, namely `getCatkinLibraryPaths`. That gets rid of the warning.

(I don't know why the anonymous function was created in the first place in #49 though.)

dirk-thomas, 2018-04-13:
@mikaelarguedas How about my proposal from the previous comment?

mikaelarguedas, 2018-04-13:
Yes I had the same idea, originally :+1: , as the function is private I think that's fine to not have it in its' own private namespace. I wanted to have a simple reproducible example to submit to GCC before removing it because that should be reported upstream.

PR #113 (2018-04-15): move catkinFindLib implementation from anonymous namespace to getCatk…
…inLibraryPaths

fixes #109
closed 2018-04-15 by mikaelarguedas