ros/pluginlib #126: [ros2] memory not cleaned up (memory leak?)

Karsten1987, 2018-08-22:
We've encountered that there is a junk of memory per loaded plugin not cleaned up.
When running a [minimal example](https://github.com/bsinno/rosbag2/tree/pluginlib_leak_example) with valgrind, we see the following output:

```
==11631== 152 bytes in 1 blocks are indirectly lost in loss record 6 of 8
==11631== at 0x4C2E0EF: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==11631== by 0x528F670: class_loader::impl::AbstractMetaObjectBase::AbstractMetaObjectBase(std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&, std::__cxx11::basic_string<char, std::char_traits, std::allocator > const&) (meta_object.cpp:56)
==11631== by 0x791766A: ???
==11631== by 0x79172FE: ???
==11631== by 0x791710A: ???
==11631== by 0x7916E49: ???
==11631== by 0x7916F33: ???
==11631== by 0x7916F49: ???
==11631== by 0x40106B9: call_init.part.0 (dl-init.c:72)
==11631== by 0x40107CA: call_init (dl-init.c:30)
==11631== by 0x40107CA: _dl_init (dl-init.c:120)
==11631== by 0x40158E1: dl_open_worker (dl-open.c:575)
==11631== by 0x4010563: _dl_catch_error (dl-error.c:187)
==11631==
==11631== 267 (16 direct, 251 indirect) bytes in 1 blocks are definitely lost in loss record 7 of 8
==11631== at 0x4C2E0EF: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==11631== by 0x79170F4: ???
==11631== by 0x7916E49: ???
==11631== by 0x7916F33: ???
==11631== by 0x7916F49: ???
==11631== by 0x40106B9: call_init.part.0 (dl-init.c:72)
==11631== by 0x40107CA: call_init (dl-init.c:30)
==11631== by 0x40107CA: _dl_init (dl-init.c:120)
==11631== by 0x40158E1: dl_open_worker (dl-open.c:575)
==11631== by 0x4010563: _dl_catch_error (dl-error.c:187)
==11631== by 0x4014DA8: _dl_open (dl-open.c:660)
==11631== by 0x6C78F08: dlopen_doit (dlopen.c:66)
==11631== by 0x4010563: _dl_catch_error (dl-error.c:187)
```
The original issue to this: https://github.com/ros2/rosbag2/issues/23

fyi @Martin-Idel-SI

mikaelarguedas, 2018-09-11:
@Martin-Idel-SI @Karsten1987 Is this only happening on shutdown ?
Or can you reproduce N * bytes leaked by loading and unloading the plugin N times ?

Martin-Idel-SI, 2018-09-11:
@mikaelarguedas No, it seems to only happen at shutdown.

Taking the example, I just duplicated the lines loading the plugin (here: https://github.com/bsinno/rosbag2/blob/8f1e7137e0e3dee13e9dc6b38213bfc3b62add3b/minimaldll/main.cpp#L26-30) and I still only see the one leak with the same amount of leaked memory

mikaelarguedas, 2018-09-11:
Thanks @Martin-Idel-SI for confirming

I suspect this is an optimization from Poco then, that knows the process is shutting down and there is no need to cleanup everything. I don't think there is much we can do on our side if that's the case

Karsten1987, 2018-09-11:
what's happening if you put this line in a loop (let's say 10x) and load and unload the plugin everytime?

mikaelarguedas, 2018-09-14:
@Martin-Idel-SI can you check if https://github.com/ros/class_loader/pull/103 fixes the issue ?

Karsten1987, 2018-09-17:
valgrind still reports lost bytes with the minimal example.
I believe it's less than before though.

mikaelarguedas, 2018-09-18:
Thanks @Karsten1987 for trying it!
Can you post the report when you get a chance to compare to the one in the issue description?

Karsten1987, 2018-09-18:
Sorry, should have mentioned it!

```
 ➭ valgrind --track-origins=yes --leak-check=full build/test_library/minimaldll
==24973== Memcheck, a memory error detector
==24973== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==24973== Using Valgrind-3.13.0.SVN and LibVEX; rerun with -h for copyright info
==24973== Command: build/test_library/minimaldll
==24973==
==24973== Syscall param msg->desc.port.name points to uninitialised byte(s)
==24973==    at 0x1007D334A: mach_msg_trap (in /usr/lib/system/libsystem_kernel.dylib)
==24973==    by 0x1007D2796: mach_msg (in /usr/lib/system/libsystem_kernel.dylib)
==24973==    by 0x1007CC485: task_set_special_port (in /usr/lib/system/libsystem_kernel.dylib)
==24973==    by 0x10096810E: _os_trace_create_debug_control_port (in /usr/lib/system/libsystem_trace.dylib)
==24973==    by 0x100968458: _libtrace_init (in /usr/lib/system/libsystem_trace.dylib)
==24973==    by 0x1004989DF: libSystem_initializer (in /usr/lib/libSystem.B.dylib)
==24973==    by 0x1000EDA1A: ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) (in /usr/lib/dyld)
==24973==    by 0x1000EDC1D: ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) (in /usr/lib/dyld)
==24973==    by 0x1000E94A9: ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) (in /usr/lib/dyld)
==24973==    by 0x1000E9440: ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) (in /usr/lib/dyld)
==24973==    by 0x1000E8523: ImageLoader::processInitializers(ImageLoader::LinkContext const&, unsigned int, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) (in /usr/lib/dyld)
==24973==    by 0x1000E85B8: ImageLoader::runInitializers(ImageLoader::LinkContext const&, ImageLoader::InitializerTimingList&) (in /usr/lib/dyld)
==24973==  Address 0x10496021c is on thread 1's stack
==24973==  in frame #2, created by task_set_special_port (???:)
==24973==  Uninitialised value was created by a stack allocation
==24973==    at 0x1009680A6: _os_trace_create_debug_control_port (in /usr/lib/system/libsystem_trace.dylib)
==24973==
--24973-- UNKNOWN mach_msg unhandled MACH_SEND_TRAILER option
--24973-- UNKNOWN mach_msg unhandled MACH_SEND_TRAILER option (repeated 2 times)
--24973-- UNKNOWN mach_msg unhandled MACH_SEND_TRAILER option (repeated 4 times)
Random number: 4
==24973==
==24973== HEAP SUMMARY:
==24973==     in use at exit: 40,759 bytes in 191 blocks
==24973==   total heap usage: 1,438 allocs, 1,247 frees, 223,266 bytes allocated
==24973==
==24973== 72 bytes in 3 blocks are possibly lost in loss record 36 of 69
==24973==    at 0x10016C232: calloc (vg_replace_malloc.c:714)
==24973==    by 0x1009DC846: map_images_nolock (in /usr/lib/libobjc.A.dylib)
==24973==    by 0x1009EFFE8: objc_object::sidetable_retainCount() (in /usr/lib/libobjc.A.dylib)
==24973==    by 0x1000DD03B: dyld::notifyBatchPartial(dyld_image_states, bool, char const* (*)(dyld_image_states, unsigned int, dyld_image_info const*), bool, bool) (in /usr/lib/dyld)
==24973==    by 0x1000DD255: dyld::registerObjCNotifiers(void (*)(unsigned int, char const* const*, mach_header const* const*), void (*)(char const*, mach_header const*), void (*)(char const*, mach_header const*)) (in /usr/lib/dyld)
==24973==    by 0x10062A00A: _dyld_objc_notify_register (in /usr/lib/system/libdyld.dylib)
==24973==    by 0x1009DC074: _objc_init (in /usr/lib/libobjc.A.dylib)
==24973==    by 0x1005BD68D: _os_object_init (in /usr/lib/system/libdispatch.dylib)
==24973==    by 0x1005BD63A: libdispatch_init (in /usr/lib/system/libdispatch.dylib)
==24973==    by 0x1004989D5: libSystem_initializer (in /usr/lib/libSystem.B.dylib)
==24973==    by 0x1000EDA1A: ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) (in /usr/lib/dyld)
==24973==    by 0x1000EDC1D: ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) (in /usr/lib/dyld)
==24973==
==24973== 136 (40 direct, 96 indirect) bytes in 1 blocks are definitely lost in loss record 48 of 69
==24973==    at 0x10016B861: malloc (vg_replace_malloc.c:302)
==24973==    by 0x1004C0E0D: operator new(unsigned long) (in /usr/lib/libc++abi.dylib)
==24973==    by 0x100387C70: class_loader::impl::loadLibrary(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, class_loader::ClassLoader*) (class_loader_core.cpp:455)
==24973==    by 0x10037DD38: class_loader::ClassLoader::loadLibrary() (class_loader.cpp:132)
==24973==    by 0x10037DA8D: class_loader::ClassLoader::ClassLoader(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) (class_loader.cpp:91)
==24973==    by 0x10037DDCB: class_loader::ClassLoader::ClassLoader(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) (class_loader.cpp:85)
==24973==    by 0x10039C2BE: class_loader::MultiLibraryClassLoader::loadLibrary(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) (multi_library_class_loader.cpp:96)
==24973==    by 0x10000C388: pluginlib::ClassLoader<StorageInterface>::loadLibraryForClass(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) (class_loader_imp.hpp:695)
[... cut ...]
==24973==    indirectly lost: 224 bytes in 3 blocks
==24973==      possibly lost: 72 bytes in 3 blocks
==24973==    still reachable: 22,474 bytes in 32 blocks
==24973==         suppressed: 17,933 bytes in 151 blocks
==24973== Reachable blocks (those to which a pointer was found) are not shown.
==24973== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==24973==
==24973== For counts of detected and suppressed errors, rerun with: -v
==24973== ERROR SUMMARY: 4 errors from 4 contexts (suppressed: 12 from 12)
```