ros/pluginlib #127: Load a class and pass arguments to the constructor [Question]

MohmadAyman, 2018-09-11:
I've an architecture where I've the `plugins` has to have arguments passed to its constructors, 

sth like this ` AuthorPlugin::AuthorPlugin(QWidget* parent) : SetupAssistantWidget(parent)` and the `SetupAssistantWidget` is the interface, is there a way to do that with `pluginlib`?

mikaelarguedas, 2018-09-14:
It is currently not possible to pass constructor arguments to plugins loaded by pluginlib (this is a restriction of `class_loader` the mechanism used for loading and instanciating the classes)

caioaamaral, 2020-02-03:
you could initialize `Qwidget parent` inside `SetupAssistantWidget` by creating some initialize function that receives a Qwidget* as param.

`AuthorPlugin::initialize(QWidget* parent) override;`

And then calling it just after default constructor of your plugin:

`global Qwidget* parent;
AuthorPlugiin some_plugin;
some_plugiin.initialize(parent)`

clalancette, 2021-05-14:
Given that this is a very old question, I'm going to close this out.  Please feel free to reopen if you think there is something to change here.

closed 2021-05-14 by clalancette