ros/pluginlib #133: Boilerplate Example for ROS2

davetcoleman, 2018-12-05:
Is there a simple boilerplate example of creating and using pluginlib in ROS2? I think this would help everyone in various ROS2 migrations. I know MoveIt! is full of pluginlib usage and we need more documentation on how to use it around ament.

SteveMacenski, 2019-10-22:
@davetcoleman I know this is ancient and you've probably figured it out by now, but here's some examples from navigation2 I've made

https://github.com/ros-planning/navigation2/tree/master/nav2_navfn_planner

That is a planning plugin, see Cmakelists for the magic you need for that, otherwise it looks more or less as you'd expect in ROS1.

https://github.com/ros-planning/navigation2/tree/master/nav2_navfn_planner

That is the action server hosting said plugin, its an incredibly simple server so its a good example of both pluginlib and action servers in ROS2

Does that meet the scope of your wants?

ruffsl, 2019-10-22:
You can also take a look at pluginlib usage in rosbag2, like the storage or converter plugin setup:  
* https://github.com/ros2/rosbag2/tree/master/rosbag2_storage_default_plugins
* https://github.com/ros2/rosbag2/tree/master/rosbag2_converter_default_plugins

DLu, 2025-02-07:
Closed by https://github.com/ros2/ros2_documentation/pull/1700

sloretz, 2025-02-07:
Thank you @DLu !

closed 2025-02-07 by sloretz

davetcoleman, 2025-02-11:
7 years ago, wow a lot changes!