ros/pluginlib #143: plugin_tool linking error

2ros0, 2019-02-05:
System information: 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux

Building ROS Lunar from source as directed here: http://wiki.ros.org/Installation/Source. At the step `./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release`, during compilation of `pluginlib`, the executable `plugin_tool` fails at the linking step with the following error:

> librosconsole.so: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
> librosconsole.so: undefined reference to `boost::re_detail::raise_runtime_error(std::runtime_error const&)'
> librosconsole.so: undefined reference to `boost::re_detail::put_mem_block(void*)'
> librosconsole.so: undefined reference to `boost::re_detail::perl_matcher<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<boost::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::_match_flags)'
> librosconsole.so: undefined reference to `boost::re_detail::get_mem_block()'
> librosconsole.so: undefined reference to `boost::re_detail::verify_options(unsigned int, boost::regex_constants::_match_flags)'
> librosconsole.so: undefined reference to `boost::re_detail::get_default_error_string(boost::regex_constants::error_type)'
> librosconsole.so: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
> collect2: error: ld returned 1 exit status
>

2ros0, 2019-02-05:
Seems this is more of a problem upstream and not within pluginlib. Possible error with boost version.

nuclearsandwich, 2019-02-05:
Continuing the discussion in https://github.com/ros/rosdistro/issues/20184.

closed 2019-02-06 by 2ros0