ros/pluginlib #147: pluginlib fails to build using clang compiler

AnticliMaxtic, 2019-03-26:
This issue was originally reported here  [pluginlib fails to build using clang compiler #678](https://github.com/ros2/ros2/issues/678) on https://github.com/ros2/ros2 where it was suggested that I post the issue to this repository 

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - release-latest bbf4dd1371df02b3f446dacd4bb4f6aed181db14
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - pluginlib

#### Steps to reproduce issue
Dockerfile:
```Dockerfile 
FROM ubuntu:18.04

ENV CHOOSE_ROS_DISTRO=crystal

RUN echo 'Etc/UTC' > /etc/timezone && \
    ln -s /usr/share/zoneinfo/Etc/UTC /etc/localtime && \
    apt-get update && apt-get install -q -y tzdata curl gnupg2 lsb-release \
    && curl http://repo.ros2.org/repos.key | apt-key add - \
    && sh -c 'echo "deb [arch=amd64,arm64] http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'  \   
    && rm -rf /var/lib/apt/lists/*

# Install tools
RUN apt-get update --fix-missing && apt-get install -y \
    build-essential \
    cmake \
    git \
    python3-colcon-common-extensions \
    python3-pip \
    python-rosdep \
    python3-vcstool \
    wget \
    libasio-dev \
    libtinyxml2-dev \
    clang \
    clang-tidy \
# install some pip packages needed for testing
    && python3 -m pip install -U \
    argcomplete \
    flake8 \
    flake8-blind-except \
    flake8-builtins \
    flake8-class-newline \
    flake8-comprehensions \
    flake8-deprecated \
    flake8-docstrings \
    flake8-import-order \
    flake8-quotes \
    git+https://github.com/lark-parser/lark.git@0.7b \
    pytest-repeat \
    pytest-rerunfailures \
    pytest \
    pytest-cov \
    pytest-runner \
    setuptools \ 
    && rm -rf /var/lib/apt/lists/*

#Setup ROS2 code for compile
WORKDIR /root/ros2_ws/ros2_base

# ADD ros2.repos /root/ros2_ws/ros2_base/ros2.repos
RUN wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos \
    && mkdir src \
    && vcs import src < ros2.repos \
    && rosdep init \
    && apt-get update && rosdep update \
    && rosdep install --from-paths src --ignore-src --rosdistro crystal -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 rti-connext-dds-5.3.1 urdfdom_headers" \
    && rm -rf /var/lib/apt/lists/*

ENV CC=/usr/bin/clang
ENV CXX=/usr/bin/clang++

WORKDIR /root/ros2_ws/ros2_base
RUN colcon build --cmake-args -DCMAKE_BUILD_TYPE=DEBUG --merge-install
```
```bash
docker build -f Dockerfile -t ros2source:latest .
```
#### Expected behavior
Successful build of the docker container
#### Actual behavior
```bash
[Processing: pluginlib, test_msgs]
--- stderr: pluginlib
CMakeFiles/pluginlib_unique_ptr_test.dir/test/unique_ptr_test.cpp.o: In function `pluginlib::ClassLoader<test_base::Fubar>::getPackageFromPluginXMLFilePath(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
unique_ptr_test.cpp:(.text._ZN9pluginlib11ClassLoaderIN9test_base5FubarEE31getPackageFromPluginXMLFilePathERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN9pluginlib11ClassLoaderIN9test_base5FubarEE31getPackageFromPluginXMLFilePathERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x59): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::parent_path() const'
unique_ptr_test.cpp:(.text._ZN9pluginlib11ClassLoaderIN9test_base5FubarEE31getPackageFromPluginXMLFilePathERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZN9pluginlib11ClassLoaderIN9test_base5FubarEE31getPackageFromPluginXMLFilePathERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x241): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::parent_path() const'
CMakeFiles/pluginlib_unique_ptr_test.dir/test/unique_ptr_test.cpp.o: In function `std::experimental::filesystem::v1::__cxx11::path::path<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::experimental::filesystem::v1::__cxx11::path>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
unique_ptr_test.cpp:(.text._ZNSt12experimental10filesystem2v17__cxx114pathC2INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES3_EERKT_[_ZNSt12experimental10filesystem2v17__cxx114pathC2INSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES3_EERKT_]+0x5e): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
CMakeFiles/pluginlib_unique_ptr_test.dir/test/unique_ptr_test.cpp.o: In function `std::experimental::filesystem::v1::exists(std::experimental::filesystem::v1::__cxx11::path const&)':
unique_ptr_test.cpp:(.text._ZNSt12experimental10filesystem2v16existsERKNS1_7__cxx114pathE[_ZNSt12experimental10filesystem2v16existsERKNS1_7__cxx114pathE]+0x11): undefined reference to `std::experimental::filesystem::v1::status(std::experimental::filesystem::v1::__cxx11::path const&)'
CMakeFiles/pluginlib_unique_ptr_test.dir/test/unique_ptr_test.cpp.o: In function `std::experimental::filesystem::v1::__cxx11::path::path<char [12], std::experimental::filesystem::v1::__cxx11::path>(char const (&) [12])':
unique_ptr_test.cpp:(.text._ZNSt12experimental10filesystem2v17__cxx114pathC2IA12_cS3_EERKT_[_ZNSt12experimental10filesystem2v17__cxx114pathC2IA12_cS3_EERKT_]+0x5b): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
CMakeFiles/pluginlib_unique_ptr_test.dir/test/unique_ptr_test.cpp.o: In function `std::experimental::filesystem::v1::__cxx11::path::_M_append(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
unique_ptr_test.cpp:(.text._ZNSt12experimental10filesystem2v17__cxx114path9_M_appendERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE[_ZNSt12experimental10filesystem2v17__cxx114path9_M_appendERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE]+0x9b): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
CMakeFiles/pluginlib_unique_ptr_test.dir/test/unique_ptr_test.cpp.o: In function `std::experimental::filesystem::v1::__cxx11::path::clear()':
unique_ptr_test.cpp:(.text._ZNSt12experimental10filesystem2v17__cxx114path5clearEv[_ZNSt12experimental10filesystem2v17__cxx114path5clearEv]+0x1e): undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [pluginlib_unique_ptr_test] Error 1
make[1]: *** [CMakeFiles/pluginlib_unique_ptr_test.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< pluginlib  [ Exited with code 2 ]
```
#### Additional information
```bash
root@b8f13126204c:~/ros2_ws/ros2_base# clang --version
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

pbeeson, 2019-03-27:
This essentially breaks the ROS2 build for clang, rendering the ROS2 Install documentation, which claims support for clang, out of date.

mjcarroll, 2019-03-27:
The issue is that the CMake logic introduced to pluginlib to handle Clang compatibility has not been backported to Crystal.  In the meantime, you can use the `ros2` branch, which compiles successfully using clang-6.0.

emersonknapp, 2019-03-27:
This was fixed in https://github.com/ros/pluginlib/commit/cc6af504b3bcd827ef84536c13ea7c1c678fc4e6

mjcarroll, 2019-03-27:
@emersonknapp that's correct for `master/ros2.repos` (which uses the `ros2` branch from here). The issue is that `release-latest` (which uses branch`crystal`) is being used in the above example.

emersonknapp, 2019-03-27:
Yeah, I just wanted to point out the exact commit.

My commit https://github.com/ros/pluginlib/commit/3bbc242f1cbdeb9a51375fb8941deb386dbf9f2f after that fixes the build for clang+libcxx, but is not necessary to support Clang on its own

pbeeson, 2019-03-27:
Thanks for the quick response.  This got us past our block with pluginlib.

AnticliMaxtic, 2019-03-28:
Thanks @emersonknapp  and @mjcarroll , this unblocks me as well. This is still an issue since as @pbeeson  mentioned this breaks the install documentation for clang using the current release-latest tag

mjcarroll, 2019-03-28:
Thanks all, I'm going to update the docs to reflect that compiling with Clang on Ubuntu is only possible Dashing forward.

closed 2019-03-28 by mjcarroll