ros/pluginlib #156: Coverity - Event unimplemented null return

danielwangksu, 2019-07-08:
Event unimplemented_null_return: | Assuming unimplemented function "Value" may return a null value.
Event dereference: | Dereferencing a pointer that might be "nullptr" "config->Value()" when calling "strcmp".
https://github.com/ros/pluginlib/blob/54c925eebf8fd8db0c5e7e9ea18cba1d32402909/pluginlib/include/pluginlib/class_loader_imp.hpp#L653

artivis, 2019-07-09:
This might be a false alarm since the pointer is check for [`NULL`  right above](https://github.com/ros/pluginlib/blob/54c925eebf8fd8db0c5e7e9ea18cba1d32402909/pluginlib/include/pluginlib/class_loader_imp.hpp#L647) and seems to be owned by the function.  
Or else it expects some particular pattern ? 
```cpp
std::string value = config->Value();
if (!(strcmp(value, "library") == 0 ...
```

artivis, 2019-07-09:
From this [tinyxml2 online doc](https://leethomason.github.io/tinyxml2/classtinyxml2_1_1_x_m_l_node.html#a66344989a4b436155bcda72bd6b07b82)
```cpp
const char* tinyxml2::XMLNode::Value	(	)	const
```
Thus it is not a false alarm.
```cpp
const char* value = config->Value();
if (NULL == value) {
    throw pluginlib::InvalidXMLException(
            "XML Document '" + xml_file +
            "' has an invalid Root Element. This likely means the XML is malformed or missing.");
    return;
}
if (!(strcmp(value, "library") == 0 ...
```

PR #158 (2020-02-14): Check for NULL in XMLNode::Value
fix #156.

Check for `const char* tinyxml2::XMLNode::Value() const == NULL` .
closed 2020-02-14 by nuclearsandwich