ros/pluginlib #170: windwos ros2-eloquent-debug  std::regex re(regex) running error

cjsJack, 2019-12-04:
Hi, I am using ros2-eloquent-debug in windows, using visual studio 2015, 
and I am building `image_common` package from [https://github.com/ros-perception/image_common](url) ,  and when I run `list_transport` project of `image_common` package, it failed when running `list_transports.exe`
So, I debug and find the reason is from `pluginlib` package from [https://github.com/ros/pluginlib](url)  ,there is a syntax error about `std::regex re(regex);` in split.hpp from `C:\dev\ros2-eloquent-debug\include\pluginlib\impl\split.hpp` _45_ line, and the parameter `regex of std::regex re(regex)` will become `"\\"` in windows system, because `C:\dev\ros2-eloquent-debug\include\pluginlib\impl\filesystem_helper.hpp` already define `#define CLASS_LOADER_IMPL_OS_DIRSEP '\\'`
so` std::regex re("\\")` will fail when running project. It is running error, I already test in vs2015, vs2017, vs2019, they all run error ` std::regex re("\\")`
Because give me some advice, thanks,
             jack

dirk-thomas, 2019-12-04:
ROS 2 Eloquent officially only supports Visual Studio 2019 (see https://index.ros.org/doc/ros2/Releases/Release-Eloquent-Elusor/).

The C++ compiler in VS 2015 might just be too old to understand the used syntax. Therefore I will go ahead and close this ticket.

dirk-thomas, 2019-12-04:
Reopening - I missed that you are describing a runtime error in the regex and that you see the same problem with VS 2019.

cjsJack, 2019-12-05:
I try run the same project list_transports.exe in ros2-dashing-debug of
windows, finally, it also run error.
 And also I debug, find problem is same, because of std::regex re("\\"),
 If you have some solution, please give me some advice
  Thanks
                , jack