ros/pluginlib #180: fails to build on Focal with clang/libcxx

dirk-thomas, 2020-03-21:
See https://ci.ros2.org/view/nightly/job/nightly_linux_clang_libcxx/358/consoleFull#console-section-208

rotu, 2020-03-22:
Confirmed locally:
```
--- stderr: pluginlib                                                                                                              
In file included from /opt/ros/master/src/ros/pluginlib/pluginlib/test/unique_ptr_test.cpp:32:
In file included from /opt/ros/master/src/ros/pluginlib/pluginlib/include/pluginlib/./class_loader.hpp:370:
In file included from /opt/ros/master/src/ros/pluginlib/pluginlib/include/pluginlib/./class_loader_imp.hpp:74:
/opt/ros/master/src/ros/pluginlib/pluginlib/include/pluginlib/./impl/filesystem_helper.hpp:62:35: warning: 'filesystem' is deprecated: std::experimental::filesystem has now been deprecated in favor of C++17's std::filesystem. Please stop using it and start using std::filesystem. This experimental version will be removed in LLVM 11. You can remove this warning by defining the _LIBCPP_NO_EXPERIMENTAL_DEPRECATION_WARNING_FILESYSTEM macro. [-Wdeprecated-declarations]
namespace fs = std::experimental::filesystem;
                                  ^
/usr/lib/llvm-10/bin/../include/c++/v1/experimental/filesystem:246:1: note: 'filesystem' has been explicitly marked deprecated here
_LIBCPP_BEGIN_NAMESPACE_EXPERIMENTAL_FILESYSTEM
^
/usr/lib/llvm-10/bin/../include/c++/v1/experimental/__config:46:63: note: expanded from macro '_LIBCPP_BEGIN_NAMESPACE_EXPERIMENTAL_FILESYSTEM'
    _LIBCPP_BEGIN_NAMESPACE_EXPERIMENTAL namespace filesystem _LIBCPP_DEPRECATED_EXPERIMENTAL_FILESYSTEM { \
                                                              ^
/usr/lib/llvm-10/bin/../include/c++/v1/experimental/__config:42:70: note: expanded from macro '_LIBCPP_DEPRECATED_EXPERIMENTAL_FILESYSTEM'
#   define _LIBCPP_DEPRECATED_EXPERIMENTAL_FILESYSTEM __attribute__((deprecated("std::experimental::filesystem has now been deprecated in favor of C++17's std::filesystem. Please stop using it and start using std::filesystem. This experimental version will be removed in LLVM 11. You can remove this warning by defining the _LIBCPP_NO_EXPERIMENTAL_DEPRECATION_WARNING_FILESYSTEM macro.")))
                                                                     ^
1 warning generated.
ld.lld: error: unable to find library -lc++fs
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [CMakeFiles/pluginlib_unique_ptr_test.dir/build.make:94: pluginlib_unique_ptr_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:132: CMakeFiles/pluginlib_unique_ptr_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /opt/ros/master/src/ros/pluginlib/pluginlib/test/utest.cpp:33:
In file included from /opt/ros/master/src/ros/pluginlib/pluginlib/include/pluginlib/./class_loader.hpp:370:
In file included from /opt/ros/master/src/ros/pluginlib/pluginlib/include/pluginlib/./class_loader_imp.hpp:74:
/opt/ros/master/src/ros/pluginlib/pluginlib/include/pluginlib/./impl/filesystem_helper.hpp:62:35: warning: 'filesystem' is deprecated: std::experimental::filesystem has now been deprecated in favor of C++17's std::filesystem. Please stop using it and start using std::filesystem. This experimental version will be removed in LLVM 11. You can remove this warning by defining the _LIBCPP_NO_EXPERIMENTAL_DEPRECATION_WARNING_FILESYSTEM macro. [-Wdeprecated-declarations]
namespace fs = std::experimental::filesystem;
                                  ^
/usr/lib/llvm-10/bin/../include/c++/v1/experimental/filesystem:246:1: note: 'filesystem' has been explicitly marked deprecated here
_LIBCPP_BEGIN_NAMESPACE_EXPERIMENTAL_FILESYSTEM
^
/usr/lib/llvm-10/bin/../include/c++/v1/experimental/__config:46:63: note: expanded from macro '_LIBCPP_BEGIN_NAMESPACE_EXPERIMENTAL_FILESYSTEM'
    _LIBCPP_BEGIN_NAMESPACE_EXPERIMENTAL namespace filesystem _LIBCPP_DEPRECATED_EXPERIMENTAL_FILESYSTEM { \
                                                              ^
/usr/lib/llvm-10/bin/../include/c++/v1/experimental/__config:42:70: note: expanded from macro '_LIBCPP_DEPRECATED_EXPERIMENTAL_FILESYSTEM'
#   define _LIBCPP_DEPRECATED_EXPERIMENTAL_FILESYSTEM __attribute__((deprecated("std::experimental::filesystem has now been deprecated in favor of C++17's std::filesystem. Please stop using it and start using std::filesystem. This experimental version will be removed in LLVM 11. You can remove this warning by defining the _LIBCPP_NO_EXPERIMENTAL_DEPRECATION_WARNING_FILESYSTEM macro.")))
                                                                     ^
1 warning generated.
ld.lld: error: unable to find library -lc++fs
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [CMakeFiles/pluginlib_utest.dir/build.make:94: pluginlib_utest] Error 1
make[1]: *** [CMakeFiles/Makefile2:103: CMakeFiles/pluginlib_utest.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< pluginlib	[ Exited with code 2 ]
```

nuclearsandwich, 2020-03-31:
Resolved by #183, which had much in common with the proposed fix in #181.

closed 2020-03-31 by nuclearsandwich