ros/pluginlib #216: Searching available plugins in ROS2

JaehyunShim, 2021-04-12:
In ROS1, you could do it by doing as the below link.
http://wiki.ros.org/pluginlib#pluginlib.2Fpluginlib_groovy.Querying_ROS_Package_System_For_Available_Plugins

What command line should  I use to do the same in ROS2? (rospack is not avaiable in ROS2) 

Thanks in advance!

KatoROSTeam, 2023-10-17:
Upvoting

mrunaljsarvaiya, 2023-10-24:
any luck on this?

KatoROSTeam, 2023-10-26:
Not yet

InigoMoreno, 2024-08-22:
Not exactly the same, but I found that moveit has a node to list the available plugins. `ros2 run moveit_ros_move_group list_move_group_capabilities`.

Maybe you can use the C++ code here for inspiration: https://github.com/moveit/moveit2/blob/5548ef152a9b4fdbf87551b4a4291ba7ea1883d9/moveit_ros/move_group/src/list_capabilities.cpp

PR #264 (2025-10-03): List plugins script
Fixes  #216
closed 2025-10-03 by ahcorde