ros/pluginlib #219: Rosdep signals missing tags

Serafadam, 2021-07-01:
Hi, when following Nav2 build from source instructions in my workspace https://navigation.ros.org/build_instructions/index.html
At the step when rosdep is required, an error pops out
```
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro foxy         
Error(s) in package '/opt/ros/foxy/build/pluginlib/prefix/share/test_pluginlib/package.xml':
Package 'test_pluginlib' must declare at least one maintainer
The package node must contain at least one "license" tag
```
Adding these tags fixes the error, I'm not sure whether this is expected behaviour.

tonynajjar, 2024-11-04:
This is still an issue with Jazzy

ahcorde, 2025-10-08:
This issue was solved here https://github.com/ros/pluginlib/pull/265, at least in `kilted` and `rolling`. I created two PRs for `jazzy` and `humble` 

 - https://github.com/ros/pluginlib/pull/283
 - https://github.com/ros/pluginlib/pull/282