ros/pluginlib #245: no match for 'operator==' error

k-ish0212, 2022-12-27:
I've encountered such an error when compiling ros2 humble. It looks rcputils::fs::path operator error but the operator seems compiled through colcon build.

> Starting >>> rcpputils
Starting >>> rosidl_runtime_c                  
--- stderr: orocos_kdl_vendor                  
Cloning into 'orocos_kdl-507de66'...
HEAD is now at 507de66 Fix CMake warning on Windows (#392)
Submodule 'python_orocos_kdl/pybind11' (https://github.com/pybind/pybind11.git) registered for path 'python_orocos_kdl/pybind11'
Cloning into '/root/xarm/ros2_humble/build/orocos_kdl_vendor/orocos_kdl-507de66-prefix/src/orocos_kdl-507de66/python_orocos_kdl/pybind11'...
---
> Finished <<< orocos_kdl_vendor [56.3s]
Starting >>> python_orocos_kdl_vendor
Finished <<< python_orocos_kdl_vendor [2.23s]  
Finished <<< rosidl_runtime_c [15.1s]           
Starting >>> rosidl_runtime_cpp
Starting >>> rosidl_generator_c
Starting >>> rmw                              
Starting >>> rosidl_typesupport_introspection_c
Finished <<< rcpputils [16.9s]                
Starting >>> rcl_logging_interface
Starting >>> libyaml_vendor
Starting >>> class_loader
Finished <<< rosidl_typesupport_introspection_c [6.88s]
Finished <<< rosidl_generator_c [10.9s]        
Finished <<< rcl_logging_interface [10.3s]     
Starting >>> rcl_logging_noop
Starting >>> rcl_logging_spdlog
Finished <<< rcl_logging_noop [1.88s]          
--- stderr: libyaml_vendor                     
Cloning into 'libyaml-0.2.5'...
HEAD is now at 2c891fc Changes for v0.2.5 release
---
> Finished <<< libyaml_vendor [13.0s]
Finished <<< rosidl_runtime_cpp [15.0s]        
Starting >>> rosidl_generator_cpp
Starting >>> rosidl_typesupport_introspection_cpp
Finished <<< class_loader [18.5s] 

> Starting >>> pluginlib
Finished <<< rosidl_typesupport_introspection_cpp [6.20s]
Finished <<< rcl_logging_spdlog [10.2s]        
Finished <<< rmw [27.9s]                       
Starting >>> rcl_yaml_param_parser
--- stderr: pluginlib                          
In file included from /root/xarm/ros2_humble/src/ros/pluginlib/pluginlib/include/pluginlib/class_loader.hpp:334,
                 from /root/xarm/ros2_humble/src/ros/pluginlib/pluginlib/test/utest.cpp:33:
/root/xarm/ros2_humble/src/ros/pluginlib/pluginlib/include/pluginlib/./class_loader_imp.hpp: In member function ‘std::string pluginlib::ClassLoader<T>::getPackageFromPluginXMLFilePath(const string&)’:
/root/xarm/ros2_humble/src/ros/pluginlib/pluginlib/include/pluginlib/./class_loader_imp.hpp:589:39: error: no match for ‘operator==’ (operand types are ‘rcpputils::fs::path’ and ‘rcpputils::fs::path’)
  589 |     if (parent.string().empty() || (p == parent)) {
      |                                     ~ ^~ ~~~~~~
      |                                     |    |
      |                                     |    rcpputils::fs::path
      |                                     rcpputils::fs::path
make[2]: *** [CMakeFiles/pluginlib_utest.dir/build.make:63: CMakeFiles/pluginlib_utest.dir/test/utest.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:187: CMakeFiles/pluginlib_utest.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
> Failed   <<< pluginlib [7.80s, exited with code 2]
Aborted  <<< rcl_yaml_param_parser [2.20s]                
Aborted  <<< rosidl_generator_cpp [24.5s]                 
Aborted  <<< fastrtps [2min 41s]                          
Aborted  <<< rviz_ogre_vendor [4min 14s]      
                                          
> Summary: 139 packages finished [4min 39s]
  1 package failed: pluginlib

Does anyone know how to solve it?
Thank you.

jdlangs, 2023-03-02:
Based on your file paths, I'm guessing you're trying to build Humble in a 20.04 container which I am doing too and I saw the exact same error. It was because I forgot to remove the Foxy versions of packages on the `AMENT_PREFIX_PATH`. So I believe `pluginlib` was compiling against the wrong version of `rcpputils`.

clalancette, 2024-10-25:
Given the latest comment here, I'm going to close this out.  If you are still having problems, please feel free to reopen.

closed 2024-10-25 by clalancette