ros/pluginlib #248: Wshadow compiler warning catches issues

Ryanf55, 2023-06-09:
Enabling Wshadow in `nav2` caught problems in pluginlib with shadowed variables. 
```
In file included from /opt/ros/rolling/include/pluginlib/pluginlib/class_loader.hpp:334,
                 from /home/ryan/Development/nav2_ws/install/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d_ros.hpp:54,
                 from /home/ryan/Development/nav2_ws/install/nav2_core/include/nav2_core/global_planner.hpp:21,
                 from /home/ryan/Development/nav2_ws/src/nav2/nav2_system_tests/src/error_codes/planner/planner_error_plugin.hpp:25,
                 from /home/ryan/Development/nav2_ws/src/nav2/nav2_system_tests/src/error_codes/planner/planner_error_plugin.cpp:15:
/opt/ros/rolling/include/pluginlib/pluginlib/./class_loader_imp.hpp: In member function 'virtual std::string pluginlib::ClassLoader<T>::getClassLibraryPath(const string&)':
/opt/ros/rolling/include/pluginlib/pluginlib/./class_loader_imp.hpp:491:13: error: declaration of 'it' shadows a previous local [-Werror=shadow]
  491 |   for (auto it = paths_to_try.begin(); it != paths_to_try.end(); it++) {
      |             ^~
/opt/ros/rolling/include/pluginlib/pluginlib/./class_loader_imp.hpp:479:20: note: shadowed declaration is here
  479 |   ClassMapIterator it = classes_available_.find(lookup_name);
  ```
  
  Branch: `rolling` from debian ` 5.3.0-1jammy.20230512.161129`
  
  I propose adding that compilation option to pluginlib `CMakeLists.txt` and fixing any shadowed variables.

SteveMacenski, 2023-10-26:
I fixed in #250

clalancette, 2023-10-27:
Fixed by #250

closed 2023-10-27 by clalancette