ros/pluginlib #259: Feature Request: Python Plugins

DLu, 2024-10-16:
It's bugged me for years that there's no official Python bindings for this package. I'd like to be able to do the following in Python. 
 * Declare an "interface" (abstract class) in a ROS package.
 * Write an implementation of that interface in another ROS package with whatever metadata to link it to the first ROS package. 
 * List all implementations of the interface across the current ROS build environment.
 * Instantiate the implementation from a string, a la `loader.createSharedInstance("polygon_plugins::Square");`

The only "mainstream" version of this that I've seen is [rqt_bag](https://github.com/ros-visualization/rqt_bag/blob/master/rqt_bag/plugin.xml) which in turn uses [`rqt_gui.RospkgPluginProvider`](https://github.com/ros-visualization/rqt/blob/ros2/rqt_gui/src/rqt_gui/rospkg_plugin_provider.py). I've also written [a different implementation](https://github.com/MetroRobots/the_navigation_gauntlet/blob/6fd2f85904cee07076956f48d2df55bc5dfbaae0/navigation_metrics/navigation_metrics/metric.py) for my own use. 

However, I think there should be one common package for this, separate from `rqt_gui`. 

Would the maintainers be open to contributions along this line?

DLu, 2024-10-16:
[image]

Relevant slide from my 2018 ROSCon talk. When all you have is a hammer....

christophebedard, 2024-10-16:
Wouldn't the equivalent in Python be the built-in plugin system with `importlib`?

DLu, 2024-10-17:
I haven't done a ton with `importlib` itself. It would definitely be used for bullet 4 above (instantiation) but is there an `importlib` equivalent to listing all plugins?

christophebedard, 2024-10-20:
I'm pretty sure, yes.

I know that `colcon` uses it for its various plugin types. For example with its `package_selection` plugin:

1. This part of `colcon_core` retrieves all `package_selection` plugins using some utility wrapper around `importlib`: https://github.com/colcon/colcon-core/blob/8a08c158bc537b734a945b3ed84f756e9627d35b/colcon_core/package_selection/__init__.py#L98
    - This calls `colcon_core.plugin_system.instantiate_extensions()`: https://github.com/colcon/colcon-core/blob/8a08c158bc537b734a945b3ed84f756e9627d35b/colcon_core/plugin_system.py#L24
    - Which calls `colcon_core.extension_point.load_extension_points()`, which uses `importlib` to return extension points along with their names for this specific extension aka "group name" (`package_selection`): https://github.com/colcon/colcon-core/blob/8a08c158bc537b734a945b3ed84f756e9627d35b/colcon_core/extension_point.py#L155
    - Base package selection class: https://github.com/colcon/colcon-core/blob/8a08c158bc537b734a945b3ed84f756e9627d35b/colcon_core/package_selection/__init__.py#L20
2. Some plugins for `package_selection` are defined the `colcon_package_selection` package: https://github.com/colcon/colcon-package-selection/blob/33f401f420cf54d9718a0593a2aec5b9748d850d/setup.cfg#L72
    - Package selection class implementation: https://github.com/colcon/colcon-package-selection/blob/33f401f420cf54d9718a0593a2aec5b9748d850d/colcon_package_selection/package_selection/dependencies.py#L34

I guess some kind of thin wrapper around `importlib` to make it simpler/closer to `pluginlib` would be a good idea.

closed 2024-10-24 by DLu